adafruit/Adafruit_WebSerial_ESPTool #3: Failure with old chrome version on Linux

jepler, 2020-12-28:
I used chromium 83.0.4103.116-1~deb10u3 because it is packaged on debian stable.  However, I ran into an error after selecting the port and speed:
```
Adafruit WebSerial ESPTool loaded.
Connecting...
Error: Failed to execute 'open' on 'SerialPort': required member baudrate is undefined
```
I didn't find any good references for this error, though other people have encountered it in various contexts, like https://stackoverflow.com/questions/64378562/web-serial-api-uncaught-in-promise-domexception-failed-to-open-serial-port and http://forum.espruino.com/comments/15634527/

oh, maybe:
```
Thanks @AkosLukacs - I was on Chrome 85 and it looks like Chrome Version 86 has broken Web Serial (they randomly renamed baudrate to baudRate!).
```
so in this case it would be a compatibility problem with old chrome versions since presumably you are using the new/correct `baudRate`.  If that's the case, this can probably be closed as unsupported.  However, maybe a version check is still worth adding?

makermelissa, 2021-01-05:
Oh yeah, I updated it to use the latest change to baudrate. Let me see if I can come up with something that can work on both.

makermelissa, 2021-01-05:
Ok fixed and tested on latest chrome and backed up to repo. Please try again at your convenience and close this if it works.

makermelissa, 2021-01-05:
Actually, I'm going to close this as I'm reasonably confident it works. We can reopen if I'm wrong. :)

closed 2021-01-05 by makermelissa