adafruit/Adafruit_WebSerial_ESPTool #4: Test out with uBlox on PyPortal Pynt

makermelissa, 2021-03-09:
I'm not sure if the uBlox module on the PyPortal Pynt has the same code as the ESP32. If not, the detection and likely new stub code will need to be added.

makermelissa, 2021-03-09:
In that case, I don't recall testing updating coprocessor firmware at all, though it should work in theory.

makermelissa, 2021-03-11:
I just tested it out and it's not working. It appears to connect momentarily, but returns a "Didn't get enough status bytes" error.
```
Adafruit WebSerial ESPTool loaded.
Connecting...
Connected successfully.
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 3000 milliseconds
Error: Didn't get enough status bytes
```

ladyada, 2021-03-11:
does it work with esptool?

makermelissa, 2021-03-11:
I've previously used ESPTool successfully in the past, but not anytime recently. Also, I tried with the WebSerial version using a regular PyPortal and had the same result.

makermelissa, 2021-03-12:
Ok, I figured out it was failing when changing baud rates. Setting it to 115200 fixes the issue. I was able to successfully program mine. I think I should add a message when it's changing baud rates so it's easier to troubleshoot in the future.

PR #6 (2021-03-12): Added baud rate change message prior to changing
Fixes #4. Also, added an else statement so it doesn't try and change it for esp8266.
closed 2021-03-12 by ladyada