adafruit/Adafruit_WebSerial_ESPTool #8: Fix changing baudrate

silbo, 2021-04-05:
### Hardware: 
- ESP WROOM 32
- CH3400 Serial IC
- Macbook Late 2013

### Software:
- Version 89.0.4389.114 (Official Build) (x86_64)
- macOS Big Sur 11.2.3

### Output:
Adafruit WebSerial ESPTool loaded.
Connecting...
Connected successfully.
Try to reset.
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Attempting to change baud rate to 460800...
Timed out after 3000 milliseconds
Error: Didn't get enough status bytes

makermelissa, 2021-04-06:
Try using 115200 if you're seeing this message.

silbo, 2021-04-06:
Yes the 115200 is the only one that works for me. Wondering if this is a limit currently with the WebSerial or we could improve the code here ?

makermelissa, 2021-04-06:
I think the code could be improved by adding a try block and falling back to 115200 if it errors and then retrying. I'm going to leave this open for now.

closed 2021-04-07 by makermelissa

conradopoole, 2021-06-18:
Just created this PR on NabuCasa's fork to get the baud rate to work above 115200

https://github.com/NabuCasa/ESP-Web-flasher/pull/13

PR #9 (2021-04-07): Make the error message clear about what the user needs to do
Fixes #8. When the app fails to connect at the higher baudrate, this makes it clear to the user about the action they need to perform to get it working.