adafruit/Adafruit_WebSerial_ESPTool #12: Garbage after flashing generic ESP8266

rafaelaroca, 2021-06-22:
Hi guys. Congratulations and thanks for Adafruit_WebSerial_ESPTool - this makes flashing ESP devices much more easier and accessible for beginners in a wide variety of environments!

I have tried to flash some ESP32 (not S2) generic boards with MicroPython firmware. All worked "nice and easy" and I quickly got MicroPython running on the boards!

My problem is with ESP8266. I have tried to flash MicroPython and CircuitPython to some ESP8266 models, and all "failed". By "failed" I mean that the Adafruit_WebSerial_ESPTool did flash the board, but when I reset the ESP8266 module, I only get continuous garbage coming from the serial port to the terminal (tried several speeds).

Here are some tests I did:

TEST 1 (works)
(with generic ESP32 / Adafruit ESPTool)

curl -LO https://github.com/nos86/micropython/releases/download/esp32-can-driver-v2/firmware.bin

On Google Chrome, select 0x1000 and click program:

Adafruit WebSerial ESPTool loaded.
Connecting...
Connected successfully.
Try to reset.
Readloop started
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Connected to ESP32
MAC Address: 24:6F:28:A1:B7:60
Uploading stub...
Running stub...
Stub is now running...
Writing data with filesize:1179600
Erase size 1179600, blocks 2304, block size 512, offset 0x1000, encrypted no
Took 126453ms to write 1179600 bytes
To run the new firmware, please reset your device.

Works at first try! :-)

By the way, if I use esptool.py, I have the output, which is consistent with the Adafruit_WebSerial_ESPTool:

Compressed 1179600 bytes to 743239...
Took 3.49s to erase flash block
Wrote 1179600 bytes (743239 compressed) at 0x00001000 in 22.9 seconds (effective 411.4 kbit/s)...
Hash of data verified.

------------

TEST 2
(Generic ESP8266 / Adafruit ESPTool)

Connecting...
Connected successfully.
Try to reset.
Readloop started
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 3000 milliseconds
Error: Didn't get enough status bytes

After reloading the webpage, works:

Adafruit WebSerial ESPTool loaded.
Connecting...
Connected successfully.
Try to reset.
Readloop started
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Connected to ESP8266EX
MAC Address: C8:2B:96:2F:43:40
Uploading stub...
Running stub...
Stub is now running...
Writing data with filesize:632064
Erase size 632064, blocks 1235, block size 512, offset 0x0000, encrypted no
Took 67159ms to write 632064 bytes
To run the new firmware, please reset your device.

After resetting, garbage is continuously sent over the serial and LED blinks randomly.

------------

TEST 3, with Generic ESP8266 / Adafruit ESPTool:

Tried to change baud rate to 460800. Chrome freezes when clicking connect, causing CPU usage to go to 100%

------------

TEST 4, with Wemos D1 mini ESP8266 / Adafruit ESPTool:

Adafruit WebSerial ESPTool loaded.
Connecting...
Connected successfully.
Try to reset.
Readloop started
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Connected to ESP8266EX
MAC Address: 2C:F4:32:3C:E5:75
Uploading stub...
Running stub...
Stub is now running...
Writing data with filesize:632064
Erase size 632064, blocks 1235, block size 512, offset 0x0000, encrypted no
Took 68114ms to write 632064 bytes
To run the new firmware, please reset your device.

After resetting, garbage is continuously sent over the serial and LED blinks randomly.

BUT...
If I flash the same file with esptool.py, it works (see below)!

esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash --flash_size=detect -fm dio 0 *bin

Erasing flash...
esptool.py v1.3
Connecting....
Running Cesanta flasher stub...
Erasing flash (this may take a while)...
Erase took 10.0 seconds
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0240
Wrote 634880 bytes at 0x0 in 55.1 seconds (92.2 kbit/s)...
Leaving...

Note that with ESP32, the bytes written are equal on esptool.py and WebSerialESP_Tool (1179600) while for ESP8266 they differ (632064 versus 634880).

Maybe I am doing something wrong, so any help / hint is welcome.

By the way, I am one of the authors of BIPES project, which allows quick and easy block based programming using serial or webrepl for MicroPython and CircuitPython. We want to allow users to flash CircuitPython or MicroPython firmwares easily and then use BIPES to program / monitor / interact with the board all from the browser.

http://bipes.net.br/beta2/ui/

Thanks for your time and best regards!

Rafael

makermelissa, 2021-06-22:
First of all, thanks for testing. This has mostly been tested with ESP32 and ESP32S2 and is not a full implementation of the official esptool. Did you try at 115200? Also, I noticed you provided `-fm dio 0`, so it's possible the issue is this is not implemented.

rafaelaroca, 2021-06-22:
> First of all, thanks for testing. This has mostly been tested with ESP32 and ESP32S2 and is not a full implementation of the official esptool. Did you try at 115200? Also, I noticed you provided `-fm dio 0`, so it's possible the issue is this is not implemented.

Hi! Thank you very much for the quick reply!

All the tests I did were @ 115200. Just tried ESP8266 at 460800bps again, and had the same issue (flashed, but flash content is inconsistent).

[...]
Baud rate can only change on ESP32 and ESP32-S2
Connected to ESP8266EX
MAC Address: C8:2B:96:2F:43:40
Uploading stub...
Running stub...
Stub is now running...
Writing data with filesize:632632
Erase size 632632, blocks 1236, block size 512, offset 0x0000, encrypted no
Took 67240ms to write 632632 bytes
To run the new firmware, please reset your device.
[...]

WebSerial_ESPTool at 460800:
Writing data with filesize:632632

WebSerial_ESPTool at 115200:
Writing data with filesize:632632

esptool.py:
Wrote 634880 bytes at 0x0 in 55.2 seconds (92.0 kbit/s)...

About -fm dio 0, it is something I have as default on my scripts, but I can also flash without -fm dio 0

esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash --flash_size=detect 0 esp8266-20191220-v1.12.bin

(Just tested the the command above works successfully and I get a MicroPython shell on my terminal)

So, DIO mode is not needed in this case.

One last point:
esptool.js has:
const ESP_SPI_SET_PARAMS = 0x0B;

esptool.py:
Flash params set to 0x0240
(which consists of:
    flash_params = struct.pack(b'BB', flash_mode, flash_size + flash_freq
on esptool.py#L3454)

I do not know if these params are related, but maybe it is something to consider.

thanks again!

Rafael

makermelissa, 2021-06-23:
> So, DIO mode is not needed in this case.
Good to know so we don't chase that red herring.

> One last point:
> esptool.js has:
> const ESP_SPI_SET_PARAMS = 0x0B;
> 
> esptool.py:
> Flash params set to 0x0240
> (which consists of:
> flash_params = struct.pack(b'BB', flash_mode, flash_size + flash_freq
> on esptool.py#L3454)
> 
> I do not know if these params are related, but maybe it is something to consider.

Yeah, likely there's some part of the esp8266 implementation that just needs to be added.

makermelissa, 2021-12-08:
I did a bunch of refactoring over the past 5 months and @brentru just tested the ESP8266 today and it worked, so I'm closing for now. We can always reopen if it's still an issue.

closed 2021-12-08 by makermelissa