adafruit/Adafruit_WebSerial_ESPTool #17: Cannot load NVM binary at overlapping address range

brentru, 2021-09-09:
This error came in through our support channel.

----

when trying to setup a huzza esp32 board. the flashing steps 1-6 complete. after the board is reset (power removed and plugged back in) it give the following error
"Error: Software loader is resident at 0x400BE000-0x400BED68. Can't load binary at overlapping address range 0x400BE000-0x400BED68. Try changing the binary loading address."

here is the log

```

Install Adafruit.io WipperSnapper for ESP32
Dark Mode
Adafruit WebSerial ESPTool loaded.
Connecting...
Connected successfully.
Try to reset.
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Connected to ESP32
MAC Address: 94:B9:7E:6B:15:E4
Uploading stub...
Running stub...
Stub is now running...
Writing data with filesize: 17168
Erase size 17168, blocks 2, block size 16384, offset 0x4096, encrypted no
50%...
100%...
Took 3007ms to write 17168 bytes
Writing data with filesize: 3072
Erase size 3072, blocks 1, block size 16384, offset 0x32768, encrypted no
Took 1438ms to write 3072 bytes
Writing data with filesize: 8192
Erase size 8192, blocks 1, block size 16384, offset 0x57344, encrypted no
Took 1430ms to write 8192 bytes
Writing data with filesize: 952112
Erase size 952112, blocks 59, block size 16384, offset 0x65536, encrypted no
1%...
3%...
5%...
6%...
8%...
10%...
11%...
13%...
15%...
16%...
18%...
20%...
22%...
23%...
25%...
27%...
28%...
30%...
32%...
33%...
35%...
37%...
38%...
40%...
42%...
44%...
45%...
47%...
49%...
50%...
52%...
54%...
55%...
57%...
59%...
61%...
62%...
64%...
66%...
67%...
69%...
71%...
72%...
74%...
76%...
77%...
79%...
81%...
83%...
84%...
86%...
88%...
89%...
91%...
93%...
94%...
96%...
98%...
100%...
Took 86932ms to write 952112 bytes
Creating NVS binary
Writing data with filesize: 12288
Erase size 12288, blocks 1, block size 16384, offset 0x9000, encrypted no
Took 1430ms to write 12288 bytes
To run the new firmware, please reset your device.
Connecting...
Connected successfully.
Try to reset.
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Connected to ESP32
MAC Address: 94:B9:7E:6B:15:E4
Uploading stub...
Error: Software loader is resident at 0x400BE000-0x400BED68. Can't load binary at overlapping address range 0x400BE000-0x400BED68. Try changing the binary loading address.```

makermelissa, 2021-09-09:
Transferred the issue, because it sounds like it's failing at the stub uploading section, which is part of the ESPTool.

kattni, 2021-12-02:
I've run into this issue as well. Here are the steps I completed before receiving it. I am running the latest Chrome, and attempting to flash an Adafruit Feather ESP32-S2 board. The board was working fine, I was simply going through the steps for guide purposes.

1. Load the WebSerial ESPTool and click "Connect"
2. Connect to the board
3. Click "Erase" and continue through successful erasure.
4. Click "Disconnect".
5. Click "Clear Text" (though I doubt this has anything to do with it).
6. Click "Connect".
7. It connects successfully, but results in the error. Full text below.
8. The button in the top left becomes "Connect" again. Clicking it again results in the same.

Full text:
```
Connecting...
Connected successfully.
Try to reset.
Connected to ESP32-S2
MAC Address: 53:97:CF:D4:B9:8B
Uploading stub...
Error: Software loader is resident at 0x40028000-0x40028FE8. Can't load binary at overlapping address range 0x40028000-0x40028FE8. Try changing the binary loading address.
```
This whole situation was unintentional - I wanted a screenshot of the text that shows up when you click "Erase", and I wasn't sure how much of what was in the traceback was actually a result of the erasure, and not already there, so I intended to erase again, and this happened instead. 

I tried refreshing Chrome, rebooting the board, reentering the bootloader, etc. Same results.

kattni, 2021-12-03:
I recovered my Feather using `esptool`, and immediately went through the steps above to try to reproduce the error. Same results. So it wasn't a one-time thing for me, it's reproducible.

makermelissa, 2021-12-06:
I found a simpler way to reproduce the issue. Connect, Disconnect, Connect Again and see error. So something isn't unloading or an additional check needs to be added.

PR #18 (2021-12-06): Fix overlapping address range error on reconnect and improve debugging
Fixes #17. The main problem was that once the Stub Loader was running, it was incorrectly detecting that it was running and attempted to run it a second time which resulted in the overlap error. This also improves debugging by making it have a couple levels of verbosity and fixes boolean debug messages.
closed 2021-12-06 by makermelissa