adafruit/Adafruit_WebSerial_ESPTool #41: Import error for esp32s3

averycooke, 2022-11-07:
Hi, this seemed like the best place to report this. 

Error: Failed to fetch dynamically imported module: https://adafruit.github.io/Adafruit_WebSerial_ESPTool/js/modules/esp32s3-b0a5f517.js

When trying to follow https://learn.adafruit.com/adafruit-qt-py-esp32-s3/factory-reset, I got the above error. 

I got things working with the python esptool, but all 3 of my ESP32-S3's that I just got seem to be missing the UF2 bootloader out of the box. Absolutely could not get them to appear as QTPYS3BOOT drive before factory reset, no trouble afterward.

makermelissa, 2022-11-07:
Thanks for letting me know. I just revived this last Friday and it's probably an issue with how I set things up.

makermelissa, 2022-11-07:
Fixed via c180f33acd99f8e3c5b5ccaf4af5fbede7e40e90

closed 2022-11-07 by makermelissa

grouma, 2022-11-08:
When will this fix land in https://adafruit.github.io/Adafruit_WebSerial_ESPTool/? I'm still experiencing the issue using that URL.

makermelissa, 2022-11-09:
Hmm, odd. I had tested it, but the problem seems to have reverted for some reason. I have a hypothesis about why this may have occurred.

makermelissa, 2022-11-09:
I think the issue is I had a GitHub action set up to upload the files to the deployment area which worked and then GitHub's page builder replaced it after. I set it to only use the action and am testing.

makermelissa, 2022-11-09:
Ok yeah, that appears to have fixed it.