adafruit/Adafruit_WebSerial_ESPTool #269: Problem with connecting to various ESP32-S3 boards

samblenny, 2024-08-28:
After a discussion on \#help-with-circuitpython, Dan suggested I file an issue here. Anyhow, I'm getting the error message below when I try to connect to 3 different ESP32-S3 boards. Same error happens with two different USB cables, using two different macs, with different versions of macOS and Chrome. The boards I tried are Adafruit Feather TFT ESP32-S3, Adafruit QT Py ESP32-S3 (8 flash / 0 psram), and Adafruit QT Py ESP32-S3 (4 flash / 2 psram).

This is the error:

```
ESP Web Flasher loaded.
Connecting...
Connected successfully.
Try hard reset.
Chip type ESP32-S3
Connected to ESP32-S3
MAC Address: [redacted]
Uploading stub...
Running stub...
[Object.debug:191] Finished read loop
Error: Invalid head of packet (0x47)
```

screenshots:
[image]

[image]

brentru, 2024-08-29:
@makermelissa  We are seeing this SlipReadError on ESP32-based boards for WipperSnapper's web flasher, too

makermelissa, 2024-08-29:
Unfortunately, I don't have an ESP32-S3 board with me at the moment. I can take a look next week when I'm back home.

timbuktwo, 2024-08-29:
I'm seeing the same issue here. 2 days ago this issue wasn't present, I think as of [this update](https://chromereleases.googleblog.com/2024/08/stable-channel-update-for-desktop_28.html), things are broken (I think). The CLI based tool still works though.

dhalbert, 2024-08-29:
@makermelissa Do you think it's worth reverting the latest release? I guess that will break other things.

dhalbert, 2024-08-29:
Maybe https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ deserves a beta site too.

makermelissa, 2024-08-29:
> @makermelissa Do you think it's worth reverting the latest release? I guess that will break other things.

Yeah, it'll break the ESP32-C6. I'm not sure what is a better approach, but I'm ok with either.

dhalbert, 2024-08-29:
@samblenny We made an update to add ESP32-C6 support and fix other things, and I think we don't want to revert, despite the regresion. Sorry that it broke ESP32-S3. Espressif has its own webtool which you could try: https://espressif.github.io/esptool-js/, while you wait for a fix.

samblenny, 2024-08-29:
@dhalbert Thanks for looking into it!

dhalbert, 2024-09-02:
@makermelissa A couple of people over the weekend got stuck with the current WebSerial tool for ESP32-S2 and ESP32-S3 and we referred them to the espressif tool. I tried the Espressif tool with Feather ESP32-C6 but it got a `Timeout` when trying to change the baud, but it works at 115200. Also the offset should be set to 0 for the Espressif tool.

makermelissa, 2024-09-04:
Ok. Thank you.

brentru, 2024-09-05:
@makermelissa  Getting this same issue on an ESP32-S3 Feather Reverse TFT as well.

```
Connecting...
Connected successfully.
Try hard reset.
Chip type ESP32-S3
Connected to ESP32-S3
MAC Address: 24:58:7C:DB:D7:70
Uploading stub...
Running stub...
[Object.debug:191] Finished read loop
Error: Invalid head of packet (0x47)
```

[image]

brentru, 2024-09-05:
The Espressif tool can connect and upload to the ESP32-S3 without showing this error.

[image]

makermelissa, 2024-09-06:
Looks like there was a bad stub. I have a PR to fix it. I just copied over from esptool, which conveniently uses the same format I made up for this tool.

PR #272 (2024-09-06): Fix ESP32-S3
Fixes #269. It looks like it had a bad stub for the ESP32-S3. I updated for the C6 and S2 as well, but those were up to date. Tested on my system locally. Problem was there before this change and gone after.
closed 2024-09-06 by dhalbert