adafruit/Adafruit_WebSerial_ESPTool #319: Last commit broke for S2 mini

dpkano, 2024-12-22:
This last [commit](https://github.com/adafruit/Adafruit_WebSerial_ESPTool/commit/e36902d862641ff1d0ef0b5abfb9fa0065ee10b0#diff-54728521937e00eb49236a8b8da0234bb36b320e2daf746b0447d6cbcee4903aL1) broke the [S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) workflow.
It still works when un-pining the esptol-js version:
From:
```javascript
import { ESPLoader, Transport } from "https://unpkg.com/esptool-js@0.5.1/bundle.js";
```
Back to:
```javascript
import { ESPLoader, Transport } from "https://unpkg.com/esptool-js/bundle.js";
```
I have not had time to investigate the reason why the 0.5.1does not work. I'll post it here if I figure it out.

Ntrondle, 2024-12-25:
that would be fantastic if this could be resolved !
many thanks for your support !

yousaf465, 2024-12-27:
previously this method worked.
Now it timesout on S2mini

[image]

Ntrondle, 2024-12-27:
Can confirm that it does indeed fix the issue with the s2-mini...

Thanks @dpkano !

yousaf465, 2024-12-28:
> The main difference from 0.5.1 to 0.5.3 is this:
> 
> <img alt="image" width="911" src="https://private-user-images.githubusercontent.com/8377485/398990545-8f8c40f5-7129-4790-8613-f01dc750be8a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUzNjAzMTcsIm5iZiI6MTczNTM2MDAxNywicGF0aCI6Ii84Mzc3NDg1LzM5ODk5MDU0NS04ZjhjNDBmNS03MTI5LTQ3OTAtODYxMy1mMDFkYzc1MGJlOGEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjhUMDQyNjU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzE0NmE0MjgwNmQ1NjYwYzQxMTNlZTI4Nzg5MWFlYzNjYjY3MzY5NDQ3MmMyNzAxOGU4ZGNjMDg2YjhmZTE2YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.kPrVpiJuX1ldPPtcuu597IQMutme4j6epJIF7a9df4Q">
> The version 0.5.3 introduces a new padding function.
> 
> @Ntrondle, if you upgrade to 0.5.3, you fix the upload for the S2 mini, and the users won't see the timeout message.

how to update to 0.5.3?

Ntrondle, 2024-12-28:
> how to update to 0.5.3?

Use this in the meantime the official website is updated. it is updated to 0.5.3

https://ntrondle.github.io/Adafruit_WebSerial_ESPTool/

closed 2024-12-30 by dhalbert

makermelissa, 2024-12-30:
The reason we pointed it to a specific version is because we are trying to avoid being surprised by an erroneous release. I just heard about this issue and updated it to use 0.5.3.

PR #320 (2024-12-30): Point to esptool-js 0.5.3
This should fix #319.