adafruit/Adafruit_WebSerial_ESPTool #321: Passthrough not working with 1.3.1 release

mikeysklar, 2025-01-07:
Airlift boards like the Matrix Portal M4 and PyPortal are no longer working with Adafruit WebSerial ESPTool since the 1.3.1 release. Direct ESP32-S2 are fine. This is likely due to the noReset logic not working properly. The code has changed quite a bit since I looked at it seven weeks ago. The JavaScript console errors I see when trying to connect with a Airlift board are:

I'm sure I can fix it, but wanted to log it here. Reported by [brblake in the forums](https://forums.adafruit.com/viewtopic.php?t=215840) 

[image]
[image]

StevenSlaa, 2025-01-08:
I also have the same problem, I think it has something to do with [esptool-js](https://github.com/espressif/esptool-js), which is what Adafruit is using. I've tested the npm package itself and got the same problem on esptool-js version: 0.5.3. Also I suspect its specifc to Mac with chrome or chromium based browsers.

To add some information to the discussion, I've tested it with a WEMOS LOLIN32, so probably not chip specific. When trying with the local esptool.py it works just fine:

```bash
% esptool.py -p /dev/tty.usbserial-0001 chip_id
esptool.py v4.5
Serial port /dev/tty.usbserial-0001
Connecting..........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: cc:7b:5c:34:ae:f4
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: ...
Hard resetting via RTS pin...
```

mikeysklar, 2025-01-08:
@StevenSlaa you are seeing something a little different. It might be another issue with esptool-js, but this issue is specific to the co-processor boards being hardReset() when trying to connect rather than avoiding reset as a passthrough mode. 

I see there are a few different type of resets available so I can try to customReset with NOP parameters to see if I can get the AirLift chatting with the new 0.5.3 release.

mikeysklar, 2025-01-08:
[Feature request for no_reset](https://github.com/espressif/esptool-js/issues/178) option filed with esptool-js.

makermelissa, 2025-01-08:
I noticed this over the weekend. I was able to work around it by using the command line esptool, but I'd love to get this working asap. Currently I am hacking on the esptool-js code to see if I can get something working.

makermelissa, 2025-01-08:
Ok, sort of got it working by just checking if mode is no_reset and skipping the reset step. However, I get `Error: Invalid head of packet`on first connection attempt. However, on the next attempt it seems to work fine. I'm going to continue working on tracing why this may be.

makermelissa, 2025-01-08:
I noticed in the passthrough code, it has a bit of code in the `setup()` function with a few delays. I suspect a race condition and wonder if adding a small delay if no_reset is requested would fix the issue.

makermelissa, 2025-01-09:
Ok, after playing around with it a bit, I found adding a delay in esptool-js got rid of the error consistently. However, there's no reason to code an error in esptool-js itself and I didn't want to rely on somebody not removing it in the future.

I played around with it further and by connecting to the passthrough board, disconnecting, and waiting before the real connection, it appear to have consistently fixed the issue. I'll submit a fix to esptool-js and then we can also add my delay fix into here.

makermelissa, 2025-01-15:
Fixed by #322.

closed 2025-01-15 by makermelissa